ut-robotics/picr21-team-jordan #1: [Mechanics] - Thrower missing chassis assembly

Akustav, 2021-09-23:
The thrower should also be attached to the chassis to better judge if it is fitting into the robot.

Akustav, 2021-09-25:
Based on the model I think the thrower motor is a little too low. When pulling the ball it, the robot can be raised up to 9mm. I would lift the motor around 6 mm and extend the C-shaped ramp to bring it closer to the ball that will be pulled into the thrower.

Image illustrates the balls motion:
[image]

VORPENV, 2021-09-25:
Done

Akustav, 2021-09-27:
Accidentally put a comment on another task but there is an issue with the internal ramp:

[image]

Maybe the sketch is a little off and takes a bad reference

VORPENV, 2021-09-27:
The sketch is changed

Akustav, 2021-09-28:
Good. I would change one thing and that is to add a chamfer to the front edge. This reduces the chance that something gets impaled while driving around: 
[image]

VORPENV, 2021-09-29:
Done

maidlam, 2021-09-30:
Resolved

closed 2021-09-30 by maidlam