ut-robotics/picr21-team-jordan #3: [Mechanics] - Weird spacer lengths

Akustav, 2021-09-23:
Spacer sizing for these specific spacers is 11.6mm. They should most likely be 12mm
[image]

Akustav, 2021-09-25:
Quick question: How will they be fastened to each other? Currently, all spacers are female-female. Will you have a long bolt in there or something else?

VORPENV, 2021-09-25:
The middle spacers will be FF, other - FM. I don't know how exactly show it in the model

Akustav, 2021-09-27:
You can have the spacers modelled out like that, so they are FF or MM. Then it is easy to see what you want to do. If you modify the spacer models on the sides we can close this issue.

Akustav, 2021-09-28:
OK

closed 2021-09-28 by Akustav