ut-robotics/picr21-team-jordan #10: Use loops instead of repeating similar code

ReikoR, 2021-10-17:
One example that could be replaced with a loop:

https://github.com/ut-robotics/picr21-team-jordan/blob/90ed4fd66dad2126e0791d80cd730da7ee57e18f/main_folder/image_calibration.py#L49

ReikoR, 2021-10-19:
You can use `map` to create a list of variables, that can be unpacked.

Something like this:
```python
trackbar_names = ["hl", "sl", "vl", "hh", "sh", "vh", "clos1", "clos2", "dil1", "dil2"]
hl, sl, vl, hh, sh, vh, clos1, clos2, dil1, dil2 = \
    map(lambda name: cv2.getTrackbarPos(name, self.trackbar_window), trackbar_names)
```

But you probably don't need to unpack the list of variables.

ReikoR, 2021-10-31:
These parts of the code could also be improved:

https://github.com/ut-robotics/picr21-team-jordan/blob/c40053266bd363a28d650fd3adbbaea86fbb1d9f/main_folder/image_calibration.py#L73-L78

https://github.com/ut-robotics/picr21-team-jordan/blob/c40053266bd363a28d650fd3adbbaea86fbb1d9f/main_folder/move_function.py#L26-L31

Leonid-98, 2021-10-31:
@ReikoR <br/>
**1:** 
https://github.com/ut-robotics/picr21-team-jordan/blob/b0aeef8138876d4ae275fab5b78f5c8cd555717f/main_folder/move_function.py#L9-L10
https://github.com/ut-robotics/picr21-team-jordan/blob/b0aeef8138876d4ae275fab5b78f5c8cd555717f/main_folder/move_function.py#L28-L31
I don't want to use map, since map is way confusing to read, so i created function
<br/>

**2:** 
https://github.com/ut-robotics/picr21-team-jordan/blob/b0aeef8138876d4ae275fab5b78f5c8cd555717f/main_folder/image_calibration.py#L72-L74

ReikoR, 2021-10-31:
You can also use list comprehension.

```python
speed1, speed2, speed3 = [calculate_speed(angle, moving_direction, speed_limit) for angle in WHEEL_ANGLES]
```

Leonid-98, 2021-11-01:
**fixed,** <br>used enums in
https://github.com/ut-robotics/picr21-team-jordan/blob/5ba7878f3f30b542db5018d997705efe433d6058/main_folder/robot_movement.py#L29

ReikoR, 2021-11-02:
Should be good enough.

closed 2021-11-02 by ReikoR