ut-robotics/picr21-team-jordan #12: Keep GUI code separate from other code

ReikoR, 2021-10-17:
Try to move GUI updating to separate functions. You could first call image and other processing functions and after that the functions that update the GUI. Also GUI could be made optional by enabling it with a command line argument when the program is started.

Leonid-98, 2021-11-01:
**fixed**<br>
there is separate gui class

ReikoR, 2021-11-02:
This code looks weird:
https://github.com/ut-robotics/picr21-team-jordan/blob/4987ebe3df4d488476baacbc978ce16e679b80f8/main_folder/image_getter.py#L105-L107

What is the purpose of it?

Leonid-98, 2021-11-02:
> 
> 
> This code looks weird:
> 
> https://github.com/ut-robotics/picr21-team-jordan/blob/4987ebe3df4d488476baacbc978ce16e679b80f8/main_folder/image_getter.py#L105-L107
> 
> What is the purpose of it?

@ReikoR i forgot to delete this line, now its gone

closed 2021-11-02 by ReikoR