ut-robotics/picr21-team-jordan #23: Omni-motion calculation linear and rotation speeds are separate

ReikoR, 2021-10-31:
Is there a reason for not having rotation speed as a separate argument for `calculate_speed` function and not adding rotation speed straight to the linear speed?

It would be better to combine linear and rotation speeds into same calculation, since the robot needs to rotate and move linearly at the same time.

https://github.com/ut-robotics/picr21-team-jordan/blob/b0aeef8138876d4ae275fab5b78f5c8cd555717f/main_folder/move_function.py#L9-L13

Leonid-98, 2021-11-08:
@ReikoR 
https://github.com/ut-robotics/picr21-team-jordan/blob/3496028531c0474792632e1844f8edd563074892/main_folder/not_tested_yet/robot_movement.py#L36-L52

is that look better already?

Leonid-98, 2021-11-11:
@ReikoR 
fixed, its works great. But only thing with i struggle with -- somehow robot took 0 degrees as moving backward, 90 as right, 180 as forward and 270 as left. So i can't use pure atan2 function, but need to check in which sector are direction is
https://github.com/ut-robotics/picr21-team-jordan/blob/1b57ac39c533dd81c75239e4c1a0b3f9dfdc1e11/main_folder/robot_movement.py#L36-L62

ReikoR, 2021-11-12:
Code could be slightly simpler if you used radians for angles.

Something like this:
```python
WHEEL_ANGLES = [math.radians(angle) for angle in [-120, 0, 120]]

def calculate_speed(self, wheel_angle, moving_direction, speed, rotation_speed):
    return int(round(math.cos(moving_direction + wheel_angle) * speed + rotation_speed))
    
moving_direction = math.atan2(speed_y, speed_x)
```

Leonid-98, 2021-11-12:
@ReikoR 
pushed into **not tested** branch:
<br>
`WHEEL_ANGLES = [math.radians(angle) for angle in [-120, 0, 120]]`
https://github.com/ut-robotics/picr21-team-jordan/blob/12909e19789e116b6b67df07fc1329d536026121/main_folder/robot_movement.py#L36-L47

closed 2021-11-16 by ReikoR