ut-robotics/picr21-team-jordan #25: Use serial.tools.list_ports

ReikoR, 2021-10-31:
https://github.com/ut-robotics/picr21-team-jordan/blob/b0aeef8138876d4ae275fab5b78f5c8cd555717f/main_folder/move_function.py#L24

Use [serial.tools.list_ports](https://pyserial.readthedocs.io/en/latest/tools.html) and port info to find the correct serial port.

Serial port number can change and it's more reliable to find the correct port name than to hardcode it.

Leonid-98, 2021-11-01:
**fixed**<br>
https://github.com/ut-robotics/picr21-team-jordan/blob/5ba7878f3f30b542db5018d997705efe433d6058/main_folder/robot_movement.py#L8-L20

ReikoR, 2021-11-02:
It would be better if you don't use the entire `hwid` string to find the correct port. `VID:PID=0483:5740` should be what you need, because it doesn't change with different STM32 microcontrollers. `SER` should be unique to the microcontroller and will be different when you have a new or different mainboard. `LOCATION` should be the physical USB port location and will change if the mainboard is connected to different USB port.

Leonid-98, 2021-11-02:
@ReikoR 
https://github.com/ut-robotics/picr21-team-jordan/blob/9c5d97e00fe527b5110b865500421555665f3b46/main_folder/robot_movement.py#L8-L24

ReikoR, 2021-11-02:
It would be better if you declare the `serial_port` variable outside the for loop and raise an exception when serial port was not found.

For example something like this:
```python
from typing import Optional

class SerialPortNotFound(Exception):
    pass

STM_32_HWID = "USB VID:PID=0483:5740"
WHEEL_ANGLES = [120, 0, -120]

class RobotMovement:
    def __init__(self):
        serial_port: Optional[str] = None        
        ports = serial.tools.list_ports.comports()
        devices = {}
        
        for port, _, hwid in sorted(ports):
            devices[hwid] = port

        for hwid in devices.keys():
            if STM_32_HWID in hwid:
                serial_port = devices[hwid]
                break

        if serial_port is None:
            raise SerialPortNotFound("Serial port not found")
    
        self.ser = serial.Serial(serial_port, 115200)
```

Leonid-98, 2021-11-03:
@ReikoR <br>
**fixed** 
https://github.com/ut-robotics/picr21-team-jordan/blob/d943730a70d844ba020a2a85a0d25d82a156d238/main_folder/robot_movement.py#L9-L34

closed 2021-11-03 by ReikoR