ut-robotics/picr21-team-jordan #28: [Mechanics] - change way roller shafts are retained

Akustav, 2021-11-02:
I would suggest changing the way how rollers sit inside the wheel housing. Currently you have 2 semicircles that enclose the shaft but you can encounter alignment issues due to the more complex matching between 2 moving semi-circles.

I would suggest a slot that is as deep as the shafts diameter and a flat surface retains this shaft in the housing.

A picture to illustrate:
[image]

VORPENV, 2021-11-04:
Do you mean that is it better to create the slot for the roller's shaft in one "semi-wheel" ? And the second "semi-wheel" will have this flat surface?

Akustav, 2021-11-07:
A deep slot for the shaft is better than 2 half-circles retaining it. I would change the design to have a deep groove for the shaft and a flat surface that holds the roller in.

Akustav, 2021-11-11:
Seems good

closed 2021-11-11 by Akustav