ut-robotics/picr21-team-jordan #38: Incorrect maximum rotation speed limiting

ReikoR, 2021-11-21:
https://github.com/ut-robotics/picr21-team-jordan/blob/9e5311730d60695c8be9be84ec046d4377a3e32b/main_folder/state_machine.py#L64-L69

Only positive side of the speed is limited and it's done twice.

It might be useful to create separate function that limits speeds between -max_speed and max_speed.

Also converting to int is probably not necessary since it's also done in `RobotMovement.calculate_speed`.

Leonid-98, 2021-11-21:
@ReikoR  
https://github.com/ut-robotics/picr21-team-jordan/blob/c345805dfce68b1c48b533b005c4b432d3ea65f4/main_folder/state_machine.py#L64-L74

ReikoR, 2021-11-21:
https://github.com/ut-robotics/picr21-team-jordan/blob/c345805dfce68b1c48b533b005c4b432d3ea65f4/main_folder/state_machine.py#L64-L69

It's better not to change the function argument.

Something like this would be better:
```python
def limit_speed(self, speed):
    if speed < -const.MAXIMUM_SPEED:
        return -const.MAXIMUM_SPEED
    elif speed > const.MAXIMUM_SPEED:
        return const.MAXIMUM_SPEED
    
    return speed
```

Or a shorter way to write the same functionality:
```python
def limit_speed(self, speed):
    return max(min(speed, const.MAXIMUM_SPEED), -const.MAXIMUM_SPEED)
```

Leonid-98, 2021-11-22:
@ReikoR 
https://github.com/ut-robotics/picr21-team-jordan/blob/ba7bd5cd1a7eb6d9ee9443e4e3b026502ec82395/main_folder/state_machine.py#L64-L70

Leonid-98, 2021-11-22:
@ReikoR removed unnecessary int conversion

closed 2021-11-23 by ReikoR