ut-robotics/picr21-team-jordan #40: Too many typos

ReikoR, 2021-11-21:
Some examples:

https://github.com/ut-robotics/picr21-team-jordan/blob/9e5311730d60695c8be9be84ec046d4377a3e32b/main_folder/image_getter.py#L14-L19

ImageCalibraion -> ImageCalibration
apllyes -> applies
StateMahchine -> StateMachine

https://github.com/ut-robotics/picr21-team-jordan/blob/9e5311730d60695c8be9be84ec046d4377a3e32b/main_folder/image_getter.py#L59

interrurpt -> interrupt 
curent -> current

Leonid-98, 2021-11-21:
@ReikoR fixed all typos with `code spell checker` extension

ReikoR, 2021-11-21:
`class ImageCalibraion` should also be fixed.

Leonid-98, 2021-11-22:
@ReikoR 
Fixed that exact typo, everything looks ok, and extension doesen't see any typos as well

closed 2021-11-23 by ReikoR