ut-robotics/picr21-team-jordan #51: Follow uniform variable naming

Akustav, 2021-11-30:
https://github.com/ut-robotics/picr21-team-jordan/blob/2ab4613d39a6d57e361e7e64e6009ab22df023fe/main_folder/main.py#L21-L34

Other variables are defined using snake case while objects, like the RobotGui use capitalized camel case. Why is it like that. I would suggest unification, since class intances and variables are basically the same thing.

For some reading, python suggest the usage of PEP-8 as a formatting standard: https://www.python.org/dev/peps/pep-0008/

Leonid-98, 2021-12-01:
@Akustav 
fixed, i just thought that class object should be marked with capital letter in the variable name

Akustav, 2021-12-01:
There should be no difference in naming between a number and an object.

closed 2021-12-01 by Akustav