ut-robotics/picr21-team-jordan #67: ManualController is not able to start and stop the game logic

ReikoR, 2021-12-22:
One of the requirements:
> Robot game logic can be started and stopped.

https://digilabor.ut.ee/index.php/2021_PiCR_grading_and_tasks#Robot_movable_with_remote_controller

Leonid-98, 2021-12-23:
https://github.com/ut-robotics/picr21-team-jordan/blob/8fe75b70bceb830b55b95326b3cf8797ee90d8c9/main_folder/main.py#L49-L59

https://github.com/ut-robotics/picr21-team-jordan/blob/8fe75b70bceb830b55b95326b3cf8797ee90d8c9/main_folder/manual_control.py#L6-L23

ReikoR, 2021-12-26:
Both `ManualController` and `StateMachine` create `RobotMovement` instance. `RobotMovement` opens serial port. This probably does not work if serial port is opened twice.

`RobotMovement` instance should probably be created in `GameLogic` class.

Leonid-98, 2021-12-26:
https://github.com/ut-robotics/picr21-team-jordan/blob/4b359ecb63d342b3ce4a8cfa97dd382b943a8926/main_folder/main.py#L27-L35
now both` ManualController` and `StateMachine` gets same `RobotMovement` instance

Leonid-98, 2022-01-08:
@ReikoR I changed the structure. `ManualController` doesn't require `RobotMovement` object, this class now only changes motor speed as variables

closed 2022-01-09 by ReikoR