ut-robotics/picr21-team-jordan #81: USB connector needs to be closer to board edge

ReikoR, 2021-12-28:
[image]

USB connector has downwards edge, that will interfere with the PCB.

Picture from another board:

[image]

valli96, 2021-12-29:
I hope this is the correct distance from the edge

ReikoR, 2021-12-29:
3D model on the footprint would help with that.

From another board it seems that around 1.6 mm from the slot hole to board edge might be the max:

[image]

You have right now 1.85mm, which is too much:

[image]

The lines on Mechanical 13 layer are also useful:

[image]

valli96, 2021-12-31:
[image]

I hope this will be fine

closed 2021-12-31 by ReikoR