ut-robotics/picr21-team-jordan #86: Fix DRC errors

ReikoR, 2021-12-29:
[image]

[image]

ReikoR, 2021-12-31:
Try to fix USB footprint. For some reason there is hole inside the slot in original footprint. The hole can be deleted.

[image]

valli96, 2021-12-31:
[image]
If I move the USB too far to the edge, I get some DRC error

ReikoR, 2021-12-31:
This line can be aligned with board edge:

[image]

To fix Hole Size Constraint Violations, increase Maximum value in HoleSize rule:

[image]

[JLCPCB Capabilities](https://jlcpcb.com/capabilities/Capabilities):
Drill Hole Size: 0.20mm- 6.30mm

valli96, 2022-01-01:
[image]
If I aligned  the connector with the board edge, I have a distance of 2.1mm

ReikoR, 2022-01-01:
For issue #81, I measured distance from the edge of the slot, not from the center.

[image]

closed 2022-01-01 by ReikoR