ut-robotics/picr21-team-jordan #88: Move ESD protection closer to USB connector

ReikoR, 2021-12-31:
Having ESD protection closer helps to avoid ESD spikes from propagating to other components.

[image]

ReikoR, 2021-12-31:
[image]

Rotating the ESD protection component would help to improve the USB signal routing:

Something like this:

[image]

valli96, 2022-01-01:
[image]

closed 2022-01-01 by ReikoR