ut-robotics/picr21-team-jordan #89: Avoid routing signals on GND layers

ReikoR, 2021-12-31:
[image]

Splits in ground might cause electric fields to spread out, which might cause crosstalk on other signals.

Something like this would be better:

[image]

valli96, 2021-12-31:
[image]
Now it is routed through over the bottom layer

ReikoR, 2021-12-31:
Now you are splitting up +16V and leaving narrow copper for high currents.

It's better to route nSLEEP near PGND polygon:

[image]

valli96, 2022-01-01:
[image]

closed 2022-01-01 by ReikoR