ut-robotics/picr21-team-jordan #90: Gerber files and Drill files export

valli96, 2022-01-02:
Altium has the function to export both Gerber files and drill files as one file. This should also be ok for JLC PCB. 

Are the settings right?
[image]

[image]

ReikoR, 2022-01-02:
X2 should probably work. I have used regular gerber (RS-274X).

From this [KiCad forum topic](https://forum.kicad.info/t/apologies-from-atommann-an-engineer-from-jlcpcb-about-gerber-x2-files-at-jlcpcb/26847/59), it seems JLCPCB is converting X2 format to RS-274X format for the software they are using.

For layers to plot you also need Mechanical 1 for board outline. You should not need keep-out and pad master layers.

valli96, 2022-01-02:
I updated the zip file and added the Mechanical 1 layer

closed 2022-01-09 by ReikoR